atsign-foundation/at_c #14: C RSA one singular mbedtls_rsa_context

JeremyTubongbanua, 2023-06-22:
In all RSA functions, we always create a new `mbedtls_rsa_context` which holds things like the public keys and private keys to do various operations.

For optimization, we should have one singular mbedtls_rsa_context once the public key and private key have been imported.

JeremyTubongbanua, 2023-10-23:
I believe a fresh RSA context should be used each time to not reuse old RNG contexts..

Will have to investigate more on this.

JeremyTubongbanua, 2024-01-22:
This is an optimization. Until we are comfortable with functionality and can move on to "nice have optimizations", this will be back logged.

JeremyTubongbanua, 2024-02-23:
Discussion: 

@JeremyTubongbanua  - Maybe importing N E D P Q every time is costly for every RSA operation.

@XavierChanth @realvarx - think it is not costly, we will investigate later

XavierChanth, 2024-02-24:
Not that it's not costly, but that there are many ways to optimize this, so we should avoid over-optimizing early.