atsign-foundation/at_c #46: free() major bug

JeremyTubongbanua, 2023-11-02:
It is possible for free() executions to occur in most functions when the variable was never allocated.

realvarx, 2023-11-02:
It could be solved by moving all the declarations and memory allocations to a point before the `goto` sentences

JeremyTubongbanua, 2023-11-14:
Assigning 2 SP;

Work involves going through each function and moving declarations near the top of the function. 

Cases where the function is called will not be so memory efficient (as we are allocating memory that we ultimately do not use in the event of an error), but I think it would be safe to assume heavy memory intensive functions used in embedded devices will be used correctly most of the time. Some proper checking can be done on the atclient layer (the layer above atchops) to ensure input is correct before calling a memory-intensive function.

PR #97 (2024-02-02): fix: free() major bug in `atchops/` and `atclient/`
closes #46 

**- What I did**

- fixed free() major bug in `atchops/` package
- fixed free() major bug in `atclient/` package
- Also added some frees() to some functions that were not freeing variables originally

**- How I did it**

- moved all initialization to the top so that all variables exist in case of an error where they are then freed.

**- How to verify it**

running REPL:

[image]

**- Description for the changelog**
closed 2024-02-02 by JeremyTubongbanua