atsign-foundation/at_c #49: ROOT_CERT should not be hardcoded

realvarx, 2023-11-03:
### Is your feature request related to a problem? Please describe.

At this moment, one of the root server certificates is hardcoded in the `connection.h` file:
https://github.com/atsign-foundation/at_c/blob/14d2107ec6f3ed6f6d3e60a59f8489a13dffa11e/packages/atclient/include/atclient/connection.h#L9-L39

It's not so much a security problem (since they are public certificates from the root server), but rather an issue that can impede functionality at a given moment (certificates expire, and this approach requires us to keep that part of the code up to date or a `root-cert.pem` file containing the updated certificate).

### Describe the solution you'd like

One option is to request the certificate chain to the root server, validate the certificates and use them to start the connection as we currently do.

Another option is to study if it's possible to avoid handling certificates in our code (SSL libraries usually have a way to do all this process automatically).

### Describe alternatives you've considered

OpenSSL: While it may seem simpler and has good support, I don't think the change is worth it if the entire connection system with MbedTLS is already in place. Additionally, it may have adverse implications for embedded systems that I'm not familiar with (I'm just noting this as an option in case other complications unrelated to this issue arise).

JeremyTubongbanua, 2023-11-13:
MbEDTLS Docs verifying cert :
https://mbed-tls.readthedocs.io/projects/api/en/development/api/file/ssl_8h/?highlight=ssl_get_peer_cert#_CPPv425mbedtls_ssl_get_peer_certPK19mbedtls_ssl_context

cpswan, 2023-11-13:
@JeremyTubongbanua there's a prominent note in that section of the docs that points towards how to correctly verify the certificate chain back to a trusted set of CAcerts rather than using a hard coded cert. The latter approach is tolerable for quick and dirty testing (or maybe even with long expiry hand rolled certs), but absolutely doesn't fit with how we'd expect the SDK to be used.

Ultimately it could be argued that a CAcerts file is just a set of hard coded certs, and that's true. One thing that we can do here is just use the CAs that we have CAA records for with our domains, namely LetsEncrypt, Google and ZeroSSL (Sectigo).

```
$ nslookup
> set type=caa
> root.atsign.org
Server:         172.28.64.1
Address:        172.28.64.1#53

Non-authoritative answer:
root.atsign.org rdata_257 = 0 issue "letsencrypt.org"
root.atsign.org rdata_257 = 0 issue "pki.goog"
root.atsign.org rdata_257 = 0 issue "sectigo.com"
```

closed 2023-11-29 by realvarx