atsign-foundation/at_c #105: at_c: use size_t where unsigned long is and stdbool for booleans

JeremyTubongbanua, 2024-02-12:
Would like a discussion of making C99 our minimum C version (because stdbool is only supported in C99+). This would mean dropping support for ANSI C (C89/90).

The sprint points allocated to this ticket will consider two things:
1. going in and replacing `unsigned long` with `size_t` from the `stddef.h` header
2. converting from int to booleans (such as iscached and ispublic in metadata) using `bool` in `stdbool.h`

JeremyTubongbanua, 2024-02-12:
@XavierChanth @cpswan Would like your input on making C99 our minimum required version. 

[mbedtls](https://github.com/Mbed-TLS/mbedtls) has their minimum version as C99.

cpswan, 2024-02-12:
I'm OK with us doing this, but I think we need to explain WHY? This may even justify an architecture decision record (ADR).

For mbedtls I'm very comfortable with it being our default. But I think from the beginning we wanted 'batteries included but removable' so that if customers had another TLS library it wouldn't be too much upheaval to switch. Any decision regarding C99 should probably align with ensuring we don't close off options there.

XavierChanth, 2024-02-12:
There are plenty of useful features and data types added in C99, and it seems to be the baseline for support because of this, many useful data types, static keyword, snprintf for string construction, etc.

If we need to support <C99 later, we can look at doing so, but we reduce time to market trying to do so now.

JeremyTubongbanua, 2024-02-14:
If this decision document is approved, I will go ahead with refactors.

https://github.com/atsign-foundation/at_protocol/pull/134

closed 2024-03-18 by JeremyTubongbanua