atsign-foundation/at_c #107: at_c: atsign type

JeremyTubongbanua, 2024-02-12:
A typedef struct holding an atsign that ensures correct format

To be used in functions that use atsigns (for example, atkey).

JeremyTubongbanua, 2024-02-12:
Some discussion has already been made here:

@XavierChanth believes less is more in that we should keep atclient_atsign_without_at at most

@realvarx wants atsign struct to hold two things: 1 a string to the atsign, and 1 string to hold atsign wtihout prefix'd @ symbol for quick access

I believe we should have 1 string to the atsign (an atclient_atstr) then a char pointer to the character after the `@` symbol, so we can use the memory of just 1 string and also have quick access to the atsign without `@` sign

XavierChanth, 2024-02-12:
#3 makes a lot of sense to me

closed 2024-02-16 by XavierChanth