atsign-foundation/at_c #112: at_c: move uuid4 into 3rdparty

JeremyTubongbanua, 2024-02-14:
This ticket is two things:

1. Discussion on moving uuid4 into 3rd party or keeping it as a fork as we have now.

Regarding this decision, some good points have been made outside of GitHub:

@JeremyTubongbanua - we could keep things in one place
@JeremyTubongbanua - there are some changes in UUId4 (such as adding support for OpenBSD) where a commit history to the library alone can be useful. People can also use our fork of UUID4 alone instead of going through at_c.

@XavierChanth - best to maintain things in one place. uuid4 is small enough to put into 3rdparty. use forks for big libraries

@realvarx - We can also keep the forks even if we add the libraries manually to 3rdparty (and copy changes to 3rdparty every time we change something in the fork), I think that's fine. I'd go for that option

2. If we decide on moving uuid4 into 3rdparty, the sprint points allocated on this ticket will measure the work of properly closing our uuid4 fork, moving uuid4 into 3rdparty/, and fixing CMake in atchops to account for the move of the uuid4 library.

XavierChanth, 2024-02-14:
@JeremyTubongbanua raises a good point: allowing people to fork our impl of uuid4 could be important

And while downloading content is a bit annoying, I think it is less of a hassle than maintaining two copies of uuid4. So as long as we maintain the code in one place, I'm happy.

XavierChanth, 2024-02-16:
I don't think we need to do anything here

closed 2024-02-16 by XavierChanth