atsign-foundation/at_c #139: at_c: check if olen is null before setting

JeremyTubongbanua, 2024-02-26:
Some functions have an olen (output length) where the written length of the output is written to the buffer is different from the actual buffer size.

Take this function for example:

```
int atchops_base64_encode(const unsigned char *src, const unsigned long srclen, unsigned char *dst,
                          const unsigned long dstlen, unsigned long *writtenlen) {
  return mbedtls_base64_encode(dst, dstlen, writtenlen, src, srclen);
}
```

Sometimes, we want to pass NULL to `unsigned long *writtenlen` in case we already know what the output length is and we won't have to needlessly allocate a variable to receive the value we don't end up using. For instance, the base64 encoded value of a 32 byte long AES key is always 44 chars in length.

This ticket's sprint points allocates the work of:
- adding NULL checks before setting *olen in all functions that use olen parameters

JeremyTubongbanua, 2024-02-26:
@realvarx Is this a problem you would encounter?

realvarx, 2024-03-04:
> @realvarx Is this a problem you would encounter?

Yes, I've encountered that problem a few times. Also, I think it’s okay to allow the developer to choose to use or not use that variable, there may be more situations where it is not necessary

JeremyTubongbanua, 2024-05-10:
Also should add a test in base64 encode when setting `*writtenlength` parameter to NULL

PR #357 (2024-07-31): feat: allow ciphertext_len to be NULL
closes #139 

**- What I did**
- There was an if-statement that should not have been there.
- Passing NULL to ciphertext_len is completely okay.

**- Description for the changelog**
feat: allow ciphertext_len to be NULL