atsign-foundation/at_c #201: at_c: Global atsign -> atkeys lookup hashmap

XavierChanth, 2024-04-24:
@realvarx @JeremyTubongbanua Would we benefit from a global hashmap store of atsign -> atkeys?

Passing around atsign is relatively easy & low cost, but passing around clones of the atkeys creates ALOT of redundant memory, when most of the time there will only ever be one (unchanging) atkeys file used by the application.

XavierChanth, 2024-04-24:
related: #174 #186

JeremyTubongbanua, 2024-05-06:
I see where this thought originated from.

Because if we have 2 atclient instances that have authenticated with the same atSign, then we have 2 copies of the same set of atKeys saved in memory.

Is that correct?

JeremyTubongbanua, 2024-05-13:
I think the solution to this would be modifying the atclient struct to have a ptr (`atclient_atkeys *atkeys`) as opposed to a static object (`atclient_atkeys`):

```
typedef atclient {
    atclient_connection secondary_connection;
    atclient_atkeys *atkeys;
}
```

That way, when we have 2 atclient contexts like this:

```c
atclient atkeys;
atclient atclient1;
atclient monitor_conn;
// ...

atclient_monitor_pkam_auth(&monitor_conn, &atkeys1);

atclient_pkam_authenticate(&atclient1, &atkeys1);
```

Then they can both be initialized with the same set of keys