atsign-foundation/at_c #222: Example ideas: atcat & attee

XavierChanth, 2024-05-03:
Let's not just build examples, let's use this time to build cool and practical tools which not only demonstrate the feature we want to show, but also how to use the feature in a useful setting.

atcat would be similar to cat, but instead of taking a file name, it would take a `<key>@<atsign>` string. The key would become a selfkey, where the namespace is automatically added: e.g. `<key>.atfiles@<atsign>` and then it would get that key and spit the contents out to stdout.

attee would be similar to tee, but instead of taking a file name, it would take a `<key>@<atsign>` string. The key would become a selfkey, where the namespace is automatically added: e.g. `<key>.atfiles@<atsign>` and then it would take the contents from the inbound pipe, wait for the pipe to close (I guess, what else can you do to avoid writing multiple keys), write the contents to the atkey & back to stdout.

I would find myself using these alot, and they are better / cooler examples than just put/get.

XavierChanth, 2024-05-03:
@cconstab what do you think?

cconstab, 2024-05-03:
BTW you can pipe into atTalk so halve of this idea is done already😁

JeremyTubongbanua, 2024-05-13:
And atscp

XavierChanth, 2024-07-21:
Maybe we can save these ideas as examples we build for the next SDK, whatever that happens to be.

XavierChanth, 2024-09-25:
two more ideas, at_ls & at_rm
list and delete the files you've stored in the atserver.

at_rm should probably take a confirmation