atsign-foundation/at_c #284: at_c: discussion to get rid of atstr

JeremyTubongbanua, 2024-05-28:
Would like to bring up discussion on getting rid of `atclient_atstr`

For reasons:
- Creating our own string library could be overwhelming. We don't want to have to teach the programmer using the C SDK how to use our special string library when we should just be using regular C strings.
- Just gets rid of lots of confusion

For functions that return strings, instead of populating an `atclient_atstr`, we should adopt the technique of passing a double pointer, where the function is in charge of allocating the memory of the string it wants to return.

So instead of a function being called similarly to:

```c
atclient_atstr result;
atclient_atstr_init(&result);

atclient_atkey_to_string(&atkey, result.str, result.size, &result.len);

// ...

atclient_atstr_free(&result);
```

We would call it like:

```c
char *result = NULL;

atclient_atkey_to_string(&atkey, &result);

// ...

free(result);
```

JeremyTubongbanua, 2024-06-12:
@XavierChanth agrees that we should use `char *` wherever we expect a null-terminated string. Only time that a length should be passed around is when we are doing cryptographic operations

JeremyTubongbanua, 2024-06-12:
@realvarx Would like your thoughts on this

realvarx, 2024-06-12:
> @realvarx Would like your thoughts on this

I believe it is the right approach when we want to define structs like `atclient_atkey`. We are already using parameters like `olen` in many `atclient` functions to determine the bytes we have used in a buffer. Although it may be slightly less convenient for us, it will also allow us greater control over how much space we use, and users will more easily understand how to use some features of our sdk.

On the other hand, I think it can still be useful in some contexts of less relevant internal functions, which the user should never need to call or use directly.

JeremyTubongbanua, 2024-06-24:
After discussion with XC and AM, we would like to go through and eliminate atstr and go through with native char * C strings.

I will be renaming the title and allocating SP to conduct the changes. Also adding to the breaking changes tracker

JeremyTubongbanua, 2024-07-30:
Closed by #337

closed 2024-07-30 by JeremyTubongbanua