atsign-foundation/at_c #287: support auth with apkam atKeys file

gkc, 2024-05-31:
### Is your feature request related to a problem? Please describe.

apkam atKeys files contain the APKAM enrollment ID which should be included in the pkam command that the client sends. The enrollment ID is required by the atServer in order to locate the APKAM public key for this enrollment, which it then uses to verify the signature.  Currently the enrollment ID is not sent, therefore the signature verification fails.

### Describe the solution you'd like

When reading atKeys json
- check if there is an enrollment ID in the json
- if an enrollment ID is present, then use it when generating the pkam command which is sent to the atServer

JeremyTubongbanua, 2024-05-31:
I wrote the atKeys code for the C SDK, so I'd be happy to be assigned to this ticket

JeremyTubongbanua, 2024-06-10:
cc @gkc 

Assigning P2 and 2 SP

JeremyTubongbanua, 2024-08-04:
Related data: https://github.com/atsign-foundation/at_libraries/blob/d764cc4e8753da294728dbe65d697ac32fce1247/packages/at_commons/lib/src/verb/pkam_verb_builder.dart#L23

PR #400 (2024-09-18): feat: APKAM enrollmentId support and atkeys_file refactor
closes #287
closes #387
closes #393 
closes #375 
closes #155 

**- What I did**
- Added `enrollment_id` support to `atkeys_file.h/.c` and `atkeys.h/.c` 
- atclient now builds pkam verb with `enrollmentId:`, if it exists
- Refactored `atkeysfile` to `atkeys_file` to be consistent with conventions
- Refactored `atclient_pkam_authenticate_options` usage in `atclient_pkam_authenticate` and `atclient_monitor_pkam_authenticate`
  - Added functional test to ensure both `options = NULL` and predefined options work with pkam authentication

**- [... cut]
closed 2024-09-18 by XavierChanth