atsign-foundation/at_c #293: at_activate in C

XavierChanth, 2024-06-05:


JeremyTubongbanua, 2024-06-06:
Where would this project live @XavierChanth ?

Would it be in `examples/` ?

cpswan, 2024-06-06:
I'm guessing here, as @XavierChanth hasn't explained *why*...

at_activate is a crucial component of the customer journey, and if their device needs C sshnpd then it likely also needs a smaller and more portable at_activate.

So not an example really, but also not part of the SDK. Maybe we need a utils directory, or maybe we need to put it in the noports repo alongside the daemon.

XavierChanth, 2024-06-06:
The primary object for creating at_activate is to make it easier to do APKAM / atSign onboarding with the C daemon. I am hesitant to put it in examples, but maybe a `bin/` directory at the root of this repo.

cconstab, 2024-08-30:
This is now something that needs to float to the top of the ToDo's.. It will enable a number of customers to automate sshnpd and its keys on IoT devices without logging into the hardware. 

**MVP**
at_activate with APKAM creation of atKeys

cconstab, 2024-09-02:
APKAM functionality is the target.. TY

JeremyTubongbanua, 2024-09-06:
We would like to move this ticket to [NoPorts](https://github.com/atsign-foundation/noports) to make things simple for our CI when we package NoPorts zips and binaries

Would it be possible to change this ticket's repo? I don't know how to do that