atsign-foundation/at_c #317: at_c: discussion on atclient_atkey_metadata_set_*

JeremyTubongbanua, 2024-06-14:
## Issue At Hand

Some functions return `void` and `int`

This is because the `void` functions do not require `malloc`, and therefore, is impossible to return any error codes, they always succeed.

The `int` functions require a `int` error code to return because it is possible for `malloc` to fail.

## Proposed Solution

Change the `void` functions to return `int`, but always return `0`.

This is my desired solution to keep everything uniform.

For example, somebody may want to automate the setting of metadata, they can write a `typedef` that returns `int` always, or something like that.

## Appendix

```h
void atclient_atkey_metadata_set_ispublic(atclient_atkey_metadata *metadata, const bool ispublic);
void atclient_atkey_metadata_set_ishidden(atclient_atkey_metadata *metadata, const bool ishidden);
void atclient_atkey_metadata_set_iscached(atclient_atkey_metadata *metadata, const bool iscached);
void atclient_atkey_metadata_set_ttl(atclient_atkey_metadata *metadata, const long ttl);
void atclient_atkey_metadata_set_ttb(atclient_atkey_metadata *metadata, const long ttb);
void atclient_atkey_metadata_set_ttr(atclient_atkey_metadata *metadata, const long ttr);
void atclient_atkey_metadata_set_ccd(atclient_atkey_metadata *metadata, const bool ccd);
void atclient_atkey_metadata_set_isbinary(atclient_atkey_metadata *metadata, const bool isbinary);
void atclient_atkey_metadata_set_isencrypted(atclient_atkey_metadata *metadata, const bool isencrypted);
int atclient_atkey_metadata_set_datasignature(atclient_atkey_metadata *metadata, const char *datasignature,
                                              const size_t datasignaturelen);
int atclient_atkey_metadata_set_sharedkeystatus(atclient_atkey_metadata *metadata, const char *sharedkeystatus,
                                                const size_t sharedkeystatuslen);
int atclient_atkey_metadata_set_sharedkeyenc(atclient_atkey_metadata *metadata, const char *sharedkeyenc,
                                             const size_t sharedkeyenclen);
int atclient_atkey_metadata_set_pubkeyhash(atclient_atkey_metadata *metadata, const char *pubkeyhash,
                                           const size_t pubkeyhashlen);
int atclient_atkey_metadata_set_pubkeyalgo(atclient_atkey_metadata *metadata, const char *pubkeyalgo,
                                           const size_t pubkeyalgolen);
int atclient_atkey_metadata_set_encoding(atclient_atkey_metadata *metadata, const char *encoding,
                                         const size_t encodinglen);
int atclient_atkey_metadata_set_enckeyname(atclient_atkey_metadata *metadata, const char *enckeyname,
                                           const size_t enckeynamelen);
int atclient_atkey_metadata_set_encalgo(atclient_atkey_metadata *metadata, const char *encalgo,
                                        const size_t encalgolen);
int atclient_atkey_metadata_set_ivnonce(atclient_atkey_metadata *metadata, const char *ivnonce,
                                        const size_t ivnoncelen);
int atclient_atkey_metadata_set_skeenckeyname(atclient_atkey_metadata *metadata, const char *skeenckeyname,
                                              const size_t skeenckeynamelen);
int atclient_atkey_metadata_set_skeencalgo(atclient_atkey_metadata *metadata, const char *skeencalgo,
                                           const size_t skeencalgolen);
```

JeremyTubongbanua, 2024-06-14:
Would like input @realvarx @XavierChanth @gkc

JeremyTubongbanua, 2024-07-15:
I am leaning towards making them all `int` and have the functions  that don't use malloc just explicitly `return 0`.

For 2 reasons:

1. **future proof** - what if there needs to be some error checking in the future in these functions
2. **uniformity** - if somebody wants to make a generic function for all the setters, it would look something like this: `typedef int setter(atclient_atkey_metadata *metadata, void *value);`

gkc, 2024-07-15:
> I am leaning towards making them all `int` and have the functions that don't use malloc just explicitly `return 0`.
>
> For 2 reasons:
>
> 1. **future proof** - what if there needs to be some error [...]

This seems reasonable to me, I can't think of any good arguments against

JeremyTubongbanua, 2024-07-17:
From @XavierChanth 
> Not super worried about this one, but also, while changing from void to int is technically a breaking change, it’s not likely to cause any functional issues for anyone.

JeremyTubongbanua, 2024-07-17:
I have +2 for making them all `ints`. Changing title and allocating SP to this ticket.

PR #337 (2024-07-30): refactor: remove `atclient_atsign`, `atclient_atstr`, refactor a bunch of function signatures
closes #263 
closes #247
closes #267
closes #127 
closes #145 
closes #153 
closes #154 
closes #128 
closes #129 
closes #317 
closes #341 
closes #335 
closes #336
closes #172 
closes #226 
closes #235 
closes #346 
closes #342 

**- What I did**

Done as of July 12th:

- Refactored `atclient_atkey_to_string` function signature to take `char **` (originally was `char *buf, const size_t bufsize, size_t *buflen`)
- Refactored `atclient_atkey_metadata_to_protocol_str` function signature to take `char * [... cut]
closed 2024-07-30 by JeremyTubongbanua