atsign-foundation/at_c #336: at_c: discussion: use `char *` for atSigns instead of having a separate struct

JeremyTubongbanua, 2024-07-10:
We had this conversation before and just wanted to get it in writing:

We currently don't use `atclient_atsign` everywhere in our function signatures, but if we were to commit to it, it would be a lot of the following 3 lines of code:

```c
atclient_atsign atsign;
atclient_atsign_init(&atsign, "@alice");

// ...

atclient_atsign_free(&atsign);
```

[image]

I would like to propose that we eliminate the `atclient_atsign` struct completely because it is always a preliminary chore step to have to create this struct in order to pass atSign string data into a function.

realvarx, 2024-07-11:
I agree, there aren't many parts where we are still using it. If we decide to reintroduce it in the future for some reason, it shouldn't be complicated

JeremyTubongbanua, 2024-07-17:
We've got +2 for deciding on removing atclient_atsign altogether.

Will be editing the title and allocating SP.

PR #337 (2024-07-30): refactor: remove `atclient_atsign`, `atclient_atstr`, refactor a bunch of function signatures
closes #263 
closes #247
closes #267
closes #127 
closes #145 
closes #153 
closes #154 
closes #128 
closes #129 
closes #317 
closes #341 
closes #335 
closes #336
closes #172 
closes #226 
closes #235 
closes #346 
closes #342 

**- What I did**

Done as of July 12th:

- Refactored `atclient_atkey_to_string` function signature to take `char **` (originally was `char *buf, const size_t bufsize, size_t *buflen`)
- Refactored `atclient_atkey_metadata_to_protocol_str` function signature to take `char * [... cut]
closed 2024-07-30 by JeremyTubongbanua