atsign-foundation/at_c #341: at_c: discussion on removing string lengths as part of function signature

JeremyTubongbanua, 2024-07-15:
## Brief Overview

We want to remove string lengths from all function signatures and simply assume that the programmer is passing in a null-terminated string.

## Issue

Here is an example of the problem at hand:

```c
int atclient_atnotification_set_encKeyName(atclient_atnotification *notification, const char *encKeyName, const size_t encKeyNamelen);

// Frequently, this function would be called like this:

atclient_atnotification_set_encKeyName(&atnotification, "123abc", strlen("123abc"));
```

But the proposal is to remove string lengths and to always assume null-terminated strings.

```c
// So it would be called like this after this discussion is passed and approved:
atclient_atnotification_set_encKeyName(&atnotification, "123abc");
```

The reason why we want to remove string length is because it can get repetitive, tedious and an eyesore when debugging code. This function for example has at most 4 strings in its function signature which ends up being 9 function arguments. yuck!

```c
int atclient_atkey_create_publickey(atclient_atkey *atkey, const char *name, const size_t namelen, const char *sharedby,
 const size_t sharedbylen, const char *namespacestr, const size_t namespacestrlen);

// Example call

atclient_atkey_create_publickey(&atkey, "name", strlen("name"), "@jeremy", strlen("@jeremy"), "@xavier", strlen("@xavier"), "wavi", strlen("wavi");

// What we want it to be

atclient_atkey_create_publickey(&atkey, "name", "@jeremy", "@xavier", "wavi");
```

## Cons

Someone may not have the null-terminated strings on-hand and have them in something like a buffer.

For example, I could do something complicated like this

```c
char my_string[] = "name|bob|jeremy"
atclient_atkey_create_publickey(&atkey, my_string[0], 4, my_string[5], 3, my_string[9], 6); 
```

If this ticket is approved, their new implementation could be something a little more tedious like this:

```c
char my_string[] = "name|bob|jeremy";
my_string[4]='\0'
my_string[8]='\0'
my_string[15]='\0'
```

But even this doesn't work all the time. What if their string buffer was no delimited. Then they would have to do some funky string splitting.

## Alternatives

We could provide two function signatures for the caller to have the flexibility, but our code base could become quite bloated.

For Example, we could follow the strcmp/strncmp where they provide options on assuming a null-terminated string vs. providing a string length:

```c
strcmp("jeremy", "jeremy123") != 0
strncmp("jeremy", "jeremy123", 6) == 0
```

JeremyTubongbanua, 2024-07-16:
From @XavierChanth :
>My 2 cents on the con listed in that ticket: yes, we lose some flexibility with null terminated strings. But the only time where a null terminated string is actually going to limit you is if you try to hyper optimize your program. At that point, you are sacrificing a lot of maintainability and creating a surface for bugs and thus I don’t think it is very practical to worry about it. There are going to be much better and safer ways to optimize first, if someone really wants to optimize further, we can leave it to them to override the implementation with a more optimal solution for their specific scenario.

gkc, 2024-07-16:
Oh I missed this until now - +1 to @XavierChanth's comments ... null-terminated strings gets my vote for sure

JeremyTubongbanua, 2024-07-17:
I've got +2 for null-terminated strings so I will be editing the title and allocating SP.

JeremyTubongbanua, 2024-07-19:
I would like to make the decision to keep strlengths as part of the function signatures in `atchops` exclusively because there comes a lot of good situations where it is better to provide string lengths.

For example, 

```c
int atchops_rsa_sign(const atchops_rsakey_privatekey privatekey, const atchops_md_type mdtype,
                     const unsigned char *message, const size_t messagelen, unsigned char *signature)
```

It would be easy to do something like this:

```c
const char *data = "data:abcdefghijk";
atchops_rsa_sign(&privatekey, ATCHOPS_MD_SHA256, data + 5, strlen(data) - 5);
```

No string manipulation was required

Another benefit of having string lengths in atchops is that no work needs to be done because it is already implemented like this everywhere in atchops

PR #337 (2024-07-30): refactor: remove `atclient_atsign`, `atclient_atstr`, refactor a bunch of function signatures
closes #263 
closes #247
closes #267
closes #127 
closes #145 
closes #153 
closes #154 
closes #128 
closes #129 
closes #317 
closes #341 
closes #335 
closes #336
closes #172 
closes #226 
closes #235 
closes #346 
closes #342 

**- What I did**

Done as of July 12th:

- Refactored `atclient_atkey_to_string` function signature to take `char **` (originally was `char *buf, const size_t bufsize, size_t *buflen`)
- Refactored `atclient_atkey_metadata_to_protocol_str` function signature to take `char * [... cut]
closed 2024-07-30 by JeremyTubongbanua