atsign-foundation/at_c #346: at_c: discussion on moving `atchops_md_type` to `sha.h`

JeremyTubongbanua, 2024-07-19:
This is `constants.h` right now:

```h
#ifndef ATCHOPS_CONSTANTS_H
#define ATCHOPS_CONSTANTS_H

#define ATCHOPS_RNG_PERSONALIZATION "@atchops12345"

#include "mbedtls/md.h"

typedef enum atchops_md_type {
  ATCHOPS_MD_NONE = 0,
  ATCHOPS_MD_MD5,
  ATCHOPS_MD_RIPEMD160,
  ATCHOPS_MD_SHA1,
  ATCHOPS_MD_SHA224,
  ATCHOPS_MD_SHA256,
  ATCHOPS_MD_SHA384,
  ATCHOPS_MD_SHA512,
  ATCHOPS_MD_SHA3_224,
  ATCHOPS_MD_SHA3_256,
  ATCHOPS_MD_SHA3_384,
  ATCHOPS_MD_SHA3_512,
} atchops_md_type;

extern const mbedtls_md_type_t atchops_mbedtls_md_map[];

#endif

```

In the perspective of the person using our atSDK, If I wanted to use a variable like `ATCHOPS_MD_SHA256`, I would expect to have to do `#include<atchops/sha.h>` as opposed to `#include<atchops/constants.h>`

## Proposal

Move the MD stuff from `constants.h` to `sha.h`

JeremyTubongbanua, 2024-07-19:
@XavierChanth Would like your input

XavierChanth, 2024-07-22:
I think this needs some more thought first. Notice the mbedtls stuff in the header file. We should wrap the extern const map in this file into an atchops/mbedtls.h file. I think the enum can move to atchops/sha.h though.

JeremyTubongbanua, 2024-07-22:
I agree with an `atchops/mbedtls.h`

PR #337 (2024-07-30): refactor: remove `atclient_atsign`, `atclient_atstr`, refactor a bunch of function signatures
closes #263 
closes #247
closes #267
closes #127 
closes #145 
closes #153 
closes #154 
closes #128 
closes #129 
closes #317 
closes #341 
closes #335 
closes #336
closes #172 
closes #226 
closes #235 
closes #346 
closes #342 

**- What I did**

Done as of July 12th:

- Refactored `atclient_atkey_to_string` function signature to take `char **` (originally was `char *buf, const size_t bufsize, size_t *buflen`)
- Refactored `atclient_atkey_metadata_to_protocol_str` function signature to take `char * [... cut]
closed 2024-07-30 by JeremyTubongbanua