atsign-foundation/at_c #350: at_c: decision log on C atSDK conventions

JeremyTubongbanua, 2024-07-23:
# 1. Snake Case

We're using snake case instead of camel case because MbedTLS happens to be snake case. Since MbedTLS is one of our core libraries seen everywhere, we may as well follow their naming convention.

When using snake case, we separate each **word** with an underscore. For example:

Good:

```c
aes_pkam_public_key
value_encrypted_for_us_base64
```

Bad:

```c
aespkampublickey
skeencalgo
``

Some exceptions that you may find throughout the code base:

- We consider `base64` as one word because "64" isn't really a word
- When we use the `at` prefix, we do not consider it as a word which is why we have `atclient` instead of `at_client`. There are just too many ats that an extra underscore would be too much to type.

# 2. Function Signature Format

Functions should return an `int` to hand off an error code to the caller of the function. If an error is not possible, void should be used (exception on this will emphasized later).

Example of a function:

```c
int calculate_length(const atclient_atkey *atkey, int *output_length);
```

Generally, function signatures should follow this format:

```c
<return_type> <function_name>(<context>, <input,..>, <output,...>);
```

- <return_type> - generally an `int` to give an error code, or may be any other return type as long as a complex error is not possible 
- <function_name> - the name of the function (example 'foo_bar') should be in snake_case
- <context> - typically this is a struct pointer that is the core object meant to be used in the function (i.e. this function LIVES and BREATHES because of the object that the caller is passing), this should be set to `const` if it is meant to be read-only
- <input,..> - any input function arguments, any values that are not editable or should not be edited, should be set to `const`.
- <output,..> - any output function arguments, typically we have pointers or double pointers to give the caller some complex values

Here are some examples:

```c
// function signature
int lemonade(atclient *ctx, const char *value, char **commit_id);
```

```c
// usage
atclient ctx; // assume ctx is set up properly (initialized, authenticated, and yada yada)
char *commit_id = NULL;
atclient_put(&ctx, "foo_bar", &commit_id);
printf("%s\n", commit_id); // outputs "37", for example
```

## 2

cpswan, 2024-07-24:
@JeremyTubongbanua this should be added to contributing.md (perhaps also with a note in README.md pointing to it and maybe listing the sections).

If there are decisions to be made here then let's have a draft ADR (but it looks to me like everything's pretty much decided).

JeremyTubongbanua, 2024-07-24:
> @JeremyTubongbanua this should be added to contributing.md (perhaps also with a note in README.md pointing to it and maybe listing the sections).
> 
> If there are decisions to be made here then let's have a draft ADR (but it looks to me like everything's pretty much decided).

Yep, will have it properly put in Contributing.MD and update the rood README.md

I'll also have an ADR put up.

closed 2024-07-24 by JeremyTubongbanua