atsign-foundation/at_c #374: at_c: automate sha256 hash `Source code (zip)` and `Source code (tar.gz)`

JeremyTubongbanua, 2024-08-14:
[image]

Ideally these two source codes are also hashed in our C Release Automation GitHub workflow

I have added this line to the release notes

```
7ca4215a473037ca07bef362b852291b0a1cf4e975d24d373d58ae9c1df832bc v0.1.0.tar.gz
```

Where I've manually sha256sum'd it myself.

The justification of doing this work is so that projects can use us with FetchContent like this:

```cmake
include(FetchContent)

FetchContent_Declare(
  atsdk
  URL https://github.com/atsign-foundation/at_c/archive/refs/tags/v0.1.0.tar.gz
  URL_HASH SHA256=7ca4215a473037ca07bef362b852291b0a1cf4e975d24d373d58ae9c1df832bc
)
FetchContent_MakeAvailable(atsdk)
```

cpswan, 2024-08-14:
@JeremyTubongbanua FetchContent should be using the at_c tarball rather than the repo tarball

JeremyTubongbanua, 2024-08-14:
> @JeremyTubongbanua FetchContent should be using the at_c tarball rather than the repo tarball

Hmm okay

```cmake
cmake_minimum_required(VERSION 3.24)

project(1-authentication)

include(FetchContent)

FetchContent_Declare(
  atclient
  URL https://github.com/atsign-foundation/at_c/releases/download/v0.1.0/at_c-v0.1.0.tar.gz
  URL_HASH SHA256=494a4960dedc45484a0078db4c388cd592efdebde45e40bdded5c2b9c587b146
  SOURCE_SUBDIR packages/atclient
)

FetchContent_Declare(
  atlogger
  URL https://github.com/atsign-foundation/at_c/releases/download/v0.1.0/at_c-v0.1.0.tar.gz
  URL_HASH SHA256=494a4960dedc45484a0078db4c388cd592efdebde45e40bdded5c2b9c587b146
  SOURCE_SUBDIR packages/atlogger
)

FetchContent_Declare(
  atchops
  URL https://github.com/atsign-foundation/at_c/releases/download/v0.1.0/at_c-v0.1.0.tar.gz
  URL_HASH SHA256=494a4960dedc45484a0078db4c388cd592efdebde45e40bdded5c2b9c587b146
  SOURCE_SUBDIR packages/atchops
)

FetchContent_Declare(
  MbedTLS
  URL https://github.com/Mbed-TLS/mbedtls/releases/download/v3.6.0/mbedtls-3.6.0.tar.bz2
  URL_HASH SHA256=3ecf94fcfdaacafb757786a01b7538a61750ebd85c4b024f56ff8ba1490fcd38
)

fetchcontent_declare(
  cjson
  URL https://github.com/DaveGamble/cJSON/archive/refs/tags/v1.7.17.zip
  URL_HASH
  SHA256=51f3b07aece8d1786e74b951fd92556506586cb36670741b6bfb79bf5d484216 # hash for v1.7.17 .zip release source code
)

FetchContent_MakeAvailable(atclient atlogger atchops cjson MbedTLS)

add_executable(main ${CMAKE_CURRENT_LIST_DIR}/main.c)

target_link_libraries(main PRIVATE atclient atlogger atchops mbedtls mbedcrypto mbedx509 cjson)

```

I have been able to get a project going like this

closed 2024-08-14 by JeremyTubongbanua