atsign-foundation/at_c #377: at_c: change `atclient_pkam_authenticate` function signature

JeremyTubongbanua, 2024-08-15:
Our C specification states that we should have optional arguments at the end of the function signature:
[image]

To stay true to that, the atclient_pkam_authenticate function signature should be changed to the following:

```c
int atclient_pkam_authenticate(atclient *ctx, const char *atsign, const char *atserver_host, const int atserver_port,
                               const atclient_atkeys *atkeys)
```

This is how it is currently:

```c
int atclient_pkam_authenticate(atclient *ctx, const char *atserver_host, const int atserver_port,
                               const atclient_atkeys *atkeys, const char *atsign)
```

The reason for this change is because of ticket #285 , which states that `atserver_host` and `atserver_port` may soon be optional arguments. `atclient_atkeys` may also soon be an optional argument as well and the function will do all the heavy lifting for the developer.

This will be a breaking change which is why I am setting this ticket as P4 (Informational) as a reminder for when we consider doing a stable release.

cpswan, 2024-08-15:
@JeremyTubongbanua whilst we're still at v0.x.y it's fine for us to do breaking changes with just a bump to minor. Only after we do v1 do we then need to start bumping major.

JeremyTubongbanua, 2024-08-15:
> e're still at v0.x.y it's fine for us to do breaking changes with just a bump to minor. Only after we do v1 do we then need to star

Awesome! Thanks for the assist

JeremyTubongbanua, 2024-09-02:
Done in #381

closed 2024-09-02 by JeremyTubongbanua