atsign-foundation/at_c #393: at_c: `message_type` and `message_body` from `monitor.c` `static int parse_message` function point to a freed address

realvarx, 2024-09-03:
The static `parse_message` function in `monitor.c` returns `message_type` and `message_body` pointers that point to memory that is freed at the end of the function. This function creates a copy of the original parameter using strdup, which internally uses malloc to allocate memory. However, the function then frees this allocated memory (`original_copy`) before returning, leaving `message_type` and `message_body` pointing to invalid memory.

https://github.com/atsign-foundation/at_c/blob/aaf943db9c9de6b28b64ebdf9903928401ef793a/packages/atclient/src/monitor.c#L247

The solution would involve returning to the old approach (use `original` parameter, and free it outside of the method after doing memcpy of the type and body), or removing the `free(original_copy)` from the exit of the method (which could lead to a memory leak if not handled properly later).

JeremyTubongbanua, 2024-09-03:
I think I see see what is happening here

Since `original_copy` is being freed, `message_type` and `message_body` are pointers to characters in the `original_copy` string. Is that correct?

JeremyTubongbanua, 2024-09-03:
Assigned 2 SP

realvarx, 2024-09-03:
> I think I see see what is happening here
> 
> Since `original_copy` is being freed, `message_type` and `message_body` are pointers to characters in the `original_copy` string. Is that correct?

Exactly

JeremyTubongbanua, 2024-09-16:
Fix for this in #400

PR #400 (2024-09-18): feat: APKAM enrollmentId support and atkeys_file refactor
closes #287
closes #387
closes #393 
closes #375 
closes #155 

**- What I did**
- Added `enrollment_id` support to `atkeys_file.h/.c` and `atkeys.h/.c` 
- atclient now builds pkam verb with `enrollmentId:`, if it exists
- Refactored `atkeysfile` to `atkeys_file` to be consistent with conventions
- Refactored `atclient_pkam_authenticate_options` usage in `atclient_pkam_authenticate` and `atclient_monitor_pkam_authenticate`
  - Added functional test to ensure both `options = NULL` and predefined options work with pkam authentication

**- [... cut]
closed 2024-09-18 by XavierChanth