atsign-foundation/at_c #399: at_c: `atauth_enrollment` interface in `packages/atauth`

JeremyTubongbanua, 2024-09-06:
We have our own way of doing enrollment requests, approvals, etc,.

- [ ] atauth_enrollment_request - user requests for namespace access in atSign | unauthorized app requests for namespace access from authorized app
- [ ] atauth_enrollment_approve - authorized app (with atSign keys) approves the user for namespace access
- [ ] atauth_enrollment_deny - authorized app (with atSign keys) denies the user for namespace access
- [ ] atauth_enrollment_list - authorized app (with atSign keys) lists all the enrolment requests from users
- [ ] atauth_enrollment_revoke - authorized app (with atSign keys), when an enrolment request was approved once, can be taken away

srieteja, 2024-10-14:
To get at_activate over the finish line only enroll:request is required. And that has been implemented. Will take up the rest of the builders once at_activate is finished. Relevant code will be a part of the at_activate PR which is still a draft. Will link the PR once it is ready.

srieteja, 2024-10-14:
Assigning 3SP to implement the rest of the command builders.

srieteja, 2024-11-11:
Impl for enroll:request added through https://github.com/atsign-foundation/at_c/pull/423/files

srieteja, 2025-01-05:
Implmentation to send enroll requests has been implemented, which is the only one required to successfully create enrollments. Updating the Priority to P3 and carrying forward to PR103, will implement the rest when I have some capacity available since they are just "Nice to have" for now.