atsign-foundation/at_c #426: atclient_connection_send fails to send cram command to secondary server

srieteja, 2024-10-14:
### Describe the bug

I have been trying to send the cram command to the secondary server. The mbedtls_ssl_write (used by atclient_connection_send) somehow returns a valid int value but the cram command is not received on the server side. Then the program exits with 'mbedtls_ssl_read failed with exit code -26624'.

### Steps to reproduce

1. First I 'fetch the challenge from the server'
2. Then I 'generate a digest with this challenge and cram secret'
3. And then 'send the command _cram:digest/r/n'_
4. This command does not reach the server and also the existing socket is terminated

### Expected behavior

The command should go through to the server and the server should respond with 'data:success'

### Screenshots

Note: The server is using my local time but the client seems to be using UTC

Server:
[image]

Client:
[image]

### Smartphones

- Device: [e.g. iPhone6]
- OS: [e.g. iOS8.1]
- Browser [e.g. stock browser, safari]
- Version [e.g. 22]

JeremyTubongbanua, 2024-10-15:
Are you remembering to add `\r\n` at the end of the src buffer?
Also make sure not to send the null-terminator.

So sending `cram:$digest\r\n` is right, but `cram:$digest\r\n\0` might give you that error

srieteja, 2024-10-15:
Closing this ticket as this is solved. The fix will be part of the at-activate PR that will be linked to https://github.com/atsign-foundation/noports/issues/1325.

closed 2024-10-15 by srieteja