atsign-foundation/at_c #442: Compiler warnings from OpenWrt

cpswan, 2024-11-08:
### Describe the bug

When compiling C sshnpd for OpenWrt a number of warnings pop up relating to at_c:

```log
[348/366] Building C object _deps/atsdk-build/packages/atclient/CMakeFiles/atclient.dir/src/notify.c.o
/home/chris/git/git.openwrt.org/openwrt_v23.05.0_x64/build_dir/target-x86_64_musl/csshnpd-c0.2.2/_deps/atsdk-src/packages/atclient/src/notify.c: In function 'generate_cmd':
/home/chris/git/git.openwrt.org/openwrt_v23.05.0_x64/build_dir/target-x86_64_musl/csshnpd-c0.2.2/_deps/atsdk-src/packages/atclient/src/notify.c:316:50: warning: format '%llu' expects argument of type 'long long unsigned int', but argument 4 has type 'uint64_t' {aka 'long unsigned int'} [-Wformat=]
  316 |     snprintf(cmd + off, cmdsize - off, ":ttln:%llu", params->notification_expiry);
      |                                               ~~~^   ~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                  |         |
      |                                                  |         uint64_t {aka long unsigned int}
      |                                                  long long unsigned int
      |                                               %lu
At top level:
cc1: note: unrecognized command-line option '-Wno-calloc-transposed-args' may have been intended to silence earlier diagnostics
```

```log
[346/366] Building C object _deps/atsdk-build/packages/atclient/CMakeFiles/atclient.dir/src/metadata.c.o
/home/chris/git/git.openwrt.org/openwrt_v23.05.0_x64/build_dir/target-x86_64_musl/csshnpd-c0.2.2/_deps/atsdk-src/packages/atclient/src/metadata.c: In function 'atclient_atkey_metadata_to_protocol_str':
/home/chris/git/git.openwrt.org/openwrt_v23.05.0_x64/build_dir/target-x86_64_musl/csshnpd-c0.2.2/_deps/atsdk-src/packages/atclient/src/metadata.c:1215:45: warning: format '%lld' expects argument of type 'long long int', but argument 3 has type 'uint64_t' {aka 'long unsigned int'} [-Wformat=]
 1215 |     sprintf((*metadata_str) + pos, ":ttl:%lld", metadata->ttl);
      |                                          ~~~^   ~~~~~~~~~~~~~
      |                                             |           |
      |                                             |           uint64_t {aka long unsigned int}
      |                                             long long int
      |                                          %ld
/home/chris/git/git.openwrt.org/openwrt_v23.05.0_x64/build_dir/target-x86_64_musl/csshnpd-c0.2.2/_deps/atsdk-src/packages/atclient/src/metadata.c:1220:45: warning: format '%lld' expects argument of type 'long long int', but argument 3 has type 'uint64_t' {aka 'long unsigned int'} [-Wformat=]
 1220 |     sprintf((*metadata_str) + pos, ":ttb:%lld", metadata->ttb);
      |                                          ~~~^   ~~~~~~~~~~~~~
      |                                             |           |
      |                                             |           uint64_t {aka long unsigned int}
      |                                             long long int
      |                                          %ld
/home/chris/git/git.openwrt.org/openwrt_v23.05.0_x64/build_dir/target-x86_64_musl/csshnpd-c0.2.2/_deps/atsdk-src/packages/atclient/src/metadata.c:1225:45: warning: format '%lld' expects argument of type 'long long int', but argument 3 has type 'uint64_t' {aka 'long unsigned int'} [-Wformat=]
 1225 |     sprintf((*metadata_str) + pos, ":ttr:%lld", metadata->ttr);
      |                                          ~~~^   ~~~~~~~~~~~~~
      |                                             |           |
      |                                             |           uint64_t {aka long unsigned int}
      |                                             long long int
      |                                          %ld
At top level:
cc1: note: unrecognized command-line option '-Wno-calloc-transposed-args' may have been intended to silence earlier diagnostics
```

### Steps to reproduce

1. First I clone the [OpenWrt repo](https://git.openwrt.org/?p=openwrt/openwrt.git;a=summary)
2. Then I `make menuconfig` and select x86_64 target
4. And then I add the [Atsign OpenWrt packages feed](https://github.com/atsign-foundation/Atsign_OpenWRT_packages)
5. Then `make V=s package/csshnpd/compile`

### Expected behavior

Build completes with no warnings

### Smartphones

- Device: [e.g. iPhone6]
- OS: [e.g. iOS8.1]
- Browser [e.g. stock browser, safari]
- Version [e.g. 22]

XavierChanth, 2024-11-08:
Same issue, I need to use the int types to log these platform independently, will make the changes.

cpswan, 2024-11-11:
Fixed by #443

closed 2024-11-11 by cpswan