atsign-foundation/at_c #477: fix: ensure auth_cli does not overwrite existing atkeys files

srieteja, 2024-12-16:
### Describe the bug

While creating new enrollments, the auth_cli can overwrite MPKAM keys file or other enrollment atKeys files.

### Steps to reproduce

1. First I 'create an enrollment at the default atKeys filePath at /home/user/.atsign/keys/'
2. Then I 'create another enrollment without providing an atkeys file path
3. This would overwrite the file created in step 1

### Expected behavior

AuthCli should not overwrite atKeys files by default. It should check if the provided filePath already exists and only overwrite if the overwrite flag is set to true.

### Smartphones

- Device: [e.g. iPhone6]
- OS: [e.g. iOS8.1]
- Browser [e.g. stock browser, safari]
- Version [e.g. 22]

gkc, 2025-01-20:
Marking for Triage

XavierChanth, 2025-01-30:
Done in 0.4.2 release

closed 2025-01-30 by XavierChanth