atsign-foundation/at_c #480: test: functional test flakes

XavierChanth, 2024-12-19:
See https://github.com/atsign-foundation/at_c/actions/runs/12414428084/attempts/1?pr=479

These tests seem to commonly flake, needs investigation.

JeremyTubongbanua, 2024-12-25:
It looks like test #6 fails because of socket connect error code 0 (which means that the socket should be reconnected)
and test #8 fails because of a failed socket read

I think upon the merge of my PR (which introduced virtual environments), these flakes should come up less

JeremyTubongbanua, 2025-01-06:
I ran the functional tests 10 times in a row and no flakes yet.

#491 highlights a new kind of flake which is in the unit tests. I am going to close this ticket and continue work in that other ticket.

closed 2025-01-06 by JeremyTubongbanua