atsign-foundation/at_c #492: test regression: atclient_find_atserver_address has false expectations when running outside of docker ve

XavierChanth, 2025-01-03:
The `FIRST_ATSIGN` argument is dynamically injected into the test at build time, but `EXPECTED_HOST` and `EXPECTED_PORT` are hard-coded to the ve atSign's host and port, which means the test will always fail outside of the virtual env.

```
#define EXPECTED_HOST "vip.ve.atsign.zone"
#define EXPECTED_PORT 25000
```

Maybe we can:
1. Disable the test (return 0) if the `FIRST_ATSIGN` doesn't match the ve one that we expect
2. If `FIRST_ATSIGN` doesn't match the ve atSign, we replace all 3 macro arguments with a well known prod atSign and it's host & port.

JeremyTubongbanua, 2025-01-04:
I think we should go with 1

If we do 2, then we run back into the issue of flaky tests

cpswan, 2025-01-06:
Surely we can just surround them with tests so that host and port can be overridden (along with the atSign)?

JeremyTubongbanua, 2025-01-06:
> Surely we can just surround them with tests so that host and port can be overridden (along with the atSign)?

That is possible is definitely possible. In my perspective, if I'm running the functional tests locally and using prod atSigns, it would be a hassle to have to get the host and port manually myself. 

After some thought, maybe using a raw socket and parsing the response and then using that as the ground truth for the test might be an option.

What do you think?

cpswan, 2025-01-06:
>  it would be a hassle to have to get the host and port manually myself.

Not really. It's a one line atDirectory query.

I know we could lean more on atDirectory here. But there could also be test cases where a test atServer isn't even in a directory. Somebody messing around with overriding at_c functional tests should know where their test atSign is.

JeremyTubongbanua, 2025-01-06:
> For sure we could use the atDirectory, but then we need another ENV variable for overriding _that_ (whether it's VE or Prod or something else).

This is what the steps look like in our GitHub actions:

```yml
- name: Install atSDK
  run: |
  cmake -S . -B build -DATSDK_BUILD_TESTS="func" -DCMAKE_BUILD_TYPE=Debug -DFIRST_ATSIGN="\"@alice🛠\"" -DSECOND_ATSIGN="\"@bob🛠\"" -DATDIRECTORY_HOST="\"vip.ve.atsign.zone\"" -DATDIRECTORY_PORT=64
  sudo cmake --build build --target install

- name: Run Functional Tests
  run: |
  ctest --test-dir build/tests/functional_tests -VV --timeout 90
```

We specify `-DATDIRECTORY_HOST` and `-DATDIRECTORY_PORT`

So I see it as we have two options:

1. Add more mandatory flags which are `-DFIRST_ATSIGN_HOST` and `-DFIRST_ATSIGN_PORT` (or instead of two flags we have one flag - `-DFIRST_ATSIGN_DOMAIN`)
2. Maybe use a raw socket to dynamically fetch the atSign's atServer host and port that test those values against what the function ends up getting you.

Regarding option 2, this is the function that is being tested in the first place:

But doing option 2 would be testing exactly what the function is doing in the first place, which makes me kind of question the robustness of the test.

```c
int atclient_utils_find_atserver_address(const char *atdirectory_host, const int atdirectory_port, const char *atsign,
                                         char **atserver_host, int *atserver_port) {
  int ret = 1;

  atclient_connection atdirectory_conn;
  atclient_connection_init(&atdirectory_conn, ATCLIENT_CONNECTION_TYPE_ATDIRECTORY);

  const size_t recv_size = 1024;
  unsigned char recv[recv_size];
  memset(recv, 0, sizeof(unsigned char) * recv_size);
  size_t recv_len = 0;

  char *atsign_without_at_symbol = NULL;
  char *cmd = NULL;

  if ((ret = atclient_connection_connect(&atdirectory_conn, atdirectory_host, atdirectory_port)) != 0) {
    atlogger_log(TAG, ATLOGGER_LOGGING_LEVEL_ERROR, "atclient_connection_connect: %d\n", ret);
    goto exit;
  }

  if ((ret = atclient_string_utils_atsign_without_at(atsign, &atsign_without_at_symbol)) != 0) {
    atlogger_log(TAG, ATLOGGER_LOGGING_LEVEL_ERROR, "atclient_string_utils_atsign_without_at: %d\n", ret);
    goto exit;
  }

  const size_t cmd_size = (strlen(atsign_without_at_symbol)) + strlen("\n") + 1;
  cmd = malloc(sizeof(char) * cmd_size);
  if (cmd == NULL) {
    atlogger_log(TAG, ATLOGGER_LOGGING_LEVEL_ERROR, "Failed to allocate memory for cmd\n");
    ret = 1;
    goto exit;
  }
  snprintf(cmd, cmd_size, "%s\n", atsign_without_at_symbol);

  if ((ret = atclient_connection_send(&atdirectory_conn, (unsigned char *)cmd, cmd_size - 1, recv, recv_size,
                                      &recv_len)) != 0) {
    atlogger_log(TAG, ATLOGGER_LOGGING_LEVEL_ERROR, "atclient_connection_send: %d\n", ret);
    goto exit;
  }

  if (recv_len == 0) {
    atlogger_log(TAG, ATLOGGER_LOGGING_LEVEL_ERROR, "No data received from atdirectory\n");
    ret = 1;
    goto exit;
  }

  // recv has something like `228aafb0-94d3-5aa2-a3b3-e36af115480d.swarm0002.atsign.zone:6943`
  // we need to split it into host and port
  char *host = strtok((char *)recv, ":");
  char *port_str = strtok(NULL, ":");
  if (port_str == NULL) {
    atlogger_log(TAG, ATLOGGER_LOGGING_LEVEL_ERROR, "Failed to parse port from atdirectory response\n");
    ret = 1;
    goto exit;
  }

  *atserver_host = strdup(host);
  if (*atserver_host == NULL) {
    atlogger_log(TAG, ATLOGGER_LOGGING_LEVEL_ERROR, "Failed to allocate memory for atserver_host\n");
    *atserver_host = NULL;
    ret = 1;
    goto exit;
  }
  *atserver_port = atoi(port_str);

  ret = 0;
  goto exit;
exit: {
  free(atsign_without_at_symbol);
  free(cmd);
  atclient_connection_free(&atdirectory_conn);
  return ret;
}
}
```

XavierChanth, 2025-01-06:
I think either approach is fine for local. Most of the time, I’m not really worried about the atDirectory code being tested locally. We have socket tests, and unit tests for parsing. The functional test is just glueing those together.

As long as the solution is simple and doesn’t take much time.

cpswan, 2025-01-06:
> The purpose of the code under testing is to make sure that atDirectory lookups are returning the expected host / port.

So why are we manually setting HOST:PORT at all?

XavierChanth, 2025-01-06:
> OK. It feels like we've gone the long way around here. But if the purpose of setting those VARS is to check the expected output of the directory then of course they need to be set with knowledge of where the atSign is, and I don't think that's a big hassle (and we very infrequently move prod atSigns).

That's what we had before, but there was some flaking, hence the move to virtual env, which lead to the values being hard-coded.

Note that more build definition VARS is not a big deal, this is what I added `just` and the justfile for.

JeremyTubongbanua, 2025-01-07:
> OK. It feels like we've gone the long way around here. But if the purpose of setting those VARS is to check the expected output of the directory then of course they need to be set with knowledge of [...]

This was implemented in #490 

We can run the tests with a command like:

```bash
cd at_c
cmake -S . -B build \
    -DATSDK_BUILD_TESTS="func" \
    -DCMAKE_BUILD_TYPE=Debug \
    -DATDIRECTORY_HOST="\"vip.ve.atsign.zone\"" \
    -DATDIRECTORY_PORT=64 \
    -DFIRST_ATSIGN="\"@barbara🛠\"" \
    -DSECOND_ATSIGN="\"@colin🛠\"" \
    -DFIRST_ATSIGN_ATSERVER_HOST="\"vip.ve.atsign.zone\"" \
    -DFIRST_ATSIGN_ATSERVER_PORT=25002 \
    -DSECOND_ATSIGN_ATSERVER_HOST="\"vip.ve.atsign.zone\"" \
    -DSECOND_ATSIGN_ATSERVER_PORT=25004 && \
cmake --build build && \
ctest --test-dir build/tests/functional_tests -VV --timeout 90
```

cpswan, 2025-01-07:
@JeremyTubongbanua this might be a good place to use [CMakePresets.json](https://cmake.org/cmake/help/latest/manual/cmake-presets.7.html)

PR #490 (2025-01-08): test: fix flaking unit test and make functional tests more adaptable to different atsigns
closes #492 
closes #491 
closes #489 

**- What I did**

Regarding `.github/workflows/tests.yaml`:
- Added `-VV` to unit tests
- Setting the test dir to the directory instead of using working-directory to make it easier to copy and paste it when testing locally
- Added `check_docker_readiness` program that checks if virtual environment has been started and also is PKAM ready.

Regarding `atclient.c`
- Fixed a segfault possibility if err_msg is NULL, then *err_msg is a seg fault.

Regarding `test_atkeys_w [... cut]
closed 2025-01-08 by XavierChanth