atsign-foundation/at_c #512: Standalone source tarball (using CMake/CPack)

cpswan, 2025-01-14:
### Is your feature request related to a problem? Please describe.

Continuing from #511 (also various issues with OpenWrt and Yocto arising from our use of FetchContent in CMakeLists.txt

### Describe the solution you'd like

Generate a source tarball that carries all dependencies so there's no need for network access during build time.

### Describe alternatives you've considered

We could package all of our dependencies separately in the respective build system.

### Additional context

The tarball generated by Cpack from #511 contains header files and `.a` files (not `.c`) but doesn't have a CMakeLists.txt to build from them

Full listing of the tarball at [this gist](https://gist.github.com/cpswan/c011498d0dcb7908e87fa6b57ea566ad)

XavierChanth, 2025-01-14:
I'm not sure that cpack has a way of bundling the .c files rather than .h

cpswan, 2025-01-14:
Well... you got it to do something very different from when I first tried it, so progress of sorts. But I also get that we might have to start over with a different approach.

XavierChanth, 2025-01-20:
Completed in PR 103

closed 2025-01-20 by XavierChanth