atsign-foundation/at_c #533: Treat base64 as null-terminated strings

XavierChanth, 2025-01-24:
@JeremyTubongbanua would love your thoughts on this.

My overall reasoning is that in the atPlatform and at_c:
- We represent byte buffers as unsigned char* and we don't null-terminate buffers
- String are char* and they are null-terminated
- base64 has becomes some hybrid of these two styles because:
  - We use them like strings
  - mbedtls treats them as byte buffers

I think because of the fact that we use base64 to serialize raw keys into strings, the best approach going forward would be to treat them like strings. It would make usage patterns elsewhere easier to identify.

The main places we would need this change are:
- atchops_base64_encode - not null-terminated but are correct type (char *)
- atchops_rsa_key_generate_base64 - null-terminated but wrong type (unsigned char*)

JeremyTubongbanua, 2025-01-24:
I agree that we should treat base64 as char*

Let's go ahead with the changes of the two functions mentioned :)

XavierChanth, 2025-02-10:
I've made these changes in a non-breaking way, see #534 for the behavioral explanation.

closed 2025-02-10 by XavierChanth