atsign-foundation/at_c #555: fix: atserver_connection should not be disconnected during free

XavierChanth, 2025-02-07:
atserver_connection_free also closes the socket, which means that if you have multiple references to an atserver_connection, such as when you fork a process. Then the first call to free will disconnect and free as intended. Every subsequent attempt to use this connection will then fail as the program state thinks its connected when its not.

Generally also, you don't want to disconnect the socket until the last instance has been closed. These actions shouldn't be coupled together. The reason for this is the underlying memory that was cloned is pointed at the same file descriptor for the socket.

XavierChanth, 2025-05-21:
Actually it's the other way around, when we net_free, we don't want to close the socket at all (which both functions do).

The relevant scenario:

When a process is forked with an active socket, there exist two processes in which the net_context memory exists, and if we call net_free in one process, it actually closes the shared file descriptor for both. Ideally, the free function doesn't perform closure of the socket, as freeing and closing are two distinctly different actions, yet treated as one by mbedtls.

However, given that (at this current time) there is no heap allocated memory in the net_context, we can simply not call net_free in the forked process, and simply suppress any leaks. The forked process is not long lived, and this issue only ever occurs with 1-2 sockets (use case dependent), so this doesn't result in a real memory leak, just extra development & testing burden.

realvarx, 2025-05-21:
I need to double check this, but it looks like "the close() only affects the child's copy of the sockfd file descriptor"

https://stackoverflow.com/questions/5133865/parent-and-child-process-socket-close

XavierChanth, 2025-05-21:
Yes, but we are working on dynamically linking to libmbedtls, which means that we need to be forwards compatible with their APIs. If they add heap allocated memory to the `mbedtls_net_context`, then I would assume the net_free would free the heap allocated memory, and net_close would not. Thus, I don't trust it. (although, to be fair their `free` function is actually `close_and_free`, so maybe their `close` function, will also be `close_and_free`)

Additionally, the lack of a `shutdown` call in the net_close function appears to be more of a mistake, rather than intentional.

In either case, I can't safely move forward on linking to libmedtls until a maintainer addresses the issue I raised...